Opentrons/opentrons #10109: 5.1 Feedback:  deckmap is missing with the protocols with long names on the slideout of device page

reachojhaneha, 2022-04-27:
### Overview

A protocol with long name doesn't displays the deck map.

### Current Behavior

[image]

The protocol has a very large name, hence the deck map is not displayed on the choose a protocol slideout

### Expected Behavior

[image]
 
deckmap and the protocol name should be displayed

### Steps To Reproduce

Upload a protocol with large name
On the device page, choose a robot -> click on Run protocol button
Choose a protocol slideout appears -> the protocol with large name doesn't displays the deckmap

nusrat813, 2022-04-29:
@mmencarelli - can you create what the design for the following: 
-if title was >3 lines 
-file name with no space

emilywools, 2022-05-17:
I talked to @b-cooper and we want to take this approach:

items to address this issue:

- [ ] deckmap needs a min width so the size is consistent no matter how long the text is
- [ ] protocol name should be displayed as 3 lines max (if it can wrap), if it can't wrap because the name has no spaces, truncate using ellipses at 1 line; add a title attribute so a user can see the full name on hover

mmencarelli, 2022-05-17:
> I talked to @b-cooper and we want to take this approach:
>
> items to address this issue:
>
> * [ ]  deckmap needs a min width so the size is consistent no matter how long the text is
> * [ ]  protocol name [...]

Sorry @nusrat813 — still needing a system on how to keep track of github mentions. Thanks, @emilywools for following up with @b-cooper. That sounds good! Min-width size should be 76px, and 3 lines max truncating content if there's more is perfect.

koji, 2022-05-20:
[image]

closed 2022-05-24 by koji