Opentrons/opentrons #10111: 5.1 Feedback:  wifi symbol is out of the <div> in robots slideout

reachojhaneha, 2022-04-27:
### Overview

wifi symbol / usb symbol is out of the <div> on Robot slideout

### Current Behavior

[image]

### Expected Behavior

[image]

### Steps To Reproduce

Goto Protocols landing page -> Goto protocols details page -> click on the protocol card, robots slideout opens.
the Wifi symbol / usb symbol is out of the <div>

mmencarelli, 2022-05-02:
The first—text wrap with the wifi appended to the end of the 2nd line! Mocked up here
[image]

closed 2022-05-18 by vabruzzo

y3rsh, 2022-05-26:
verified
[image]

PR #10326 (2022-05-18): style(app): fix icon wrapping in robots slideout

# Overview

This will fix the icon that overflows outside the robot slideout div.
Closes #10111 

# Changelog

- Restructure `AvailableRobotOption` markup such that the icon wraps  if necessary and sits at the end of the last line of text.

# Review requests

- Pull open the robot slideout to confirm the icon acts as desired.
- Ensure that code is following conventions, in particular I suspect we might want to do something with the negative margin-bottom. I've left a comment in the PR highlighting this.

# Risk assessment
Low