Opentrons/opentrons #10121: 5.1 Feedback:  App screen goes white after turning on OT-2, clicking Devices tab

hartm9, 2022-04-28:
### Overview

I'm having the same issue while using a Windows computer. If I start the App first, it loads without issue and I can switch to different tabs, but when I turn the OT-2 on and click on the Devices tab, the App goes white and I'm unable to see or click on anything in the App.

If the OT-2 is on and I start the App, I can see the Devices tab for a split second, then
it goes back to white. Clicking Reload and Force Reload causes the Devices tab to flicker into view for a second, but then the App immediately goes back to white. When I open the Developer Tools, I can see there are a couple TypeErrors, I'll attach a screenshot to this post.
Whitescreen_dev_tools_Matthew_Hart_220426

In this state, I'm unable to use the OT-2. Any recommendations? Thanks!

### Current Behavior

App starts, OT-2 turns on, click on Devices tab, screen goes white
Or, OT-2 is on, App starts, shows Devices tab for a split second, then goes white

### Expected Behavior

Screen should not go white

### Operating system

Windows

### Robot setup or anything else?

Connected via Wi-Fi

koji, 2022-04-28:
@hartm9 seems that your screenshot wasn't uploaded properly

nusrat813, 2022-04-29:
Fix in for beta.1 - file corruption not being protected by the app 

#10069

closed 2022-04-29 by nusrat813

y3rsh, 2022-05-26:
I am not seeing this white screen behavior.