Opentrons/opentrons #10122: LL: Some product images contain a line on the right side

mikecardarelli, 2022-04-28:


koji, 2022-04-28:
@mikecardarelli @skyewindsor 
5.1.0 (I saw this when I reviewed labware PR from Sarah)
https://github.com/Opentrons/opentrons/tree/edge/app/src/organisms/Labware/LabwareDetails/images

old one?
https://github.com/Opentrons/opentrons/tree/edge/labware-library/src/images

skyewindsor, 2022-05-11:
@koji There was only one image that I could find with this cropping line issue. So, I think this ticket is fully solved by https://github.com/Opentrons/opentrons/pull/10256

koji, 2022-05-11:
@skyewindsor
If my memory serves me correctly,  we will need to move images into one folder.  #10163 
If we move images, we will need to modify the labware code. That is the reason why I said  'partially'.

closed 2022-05-12 by skyewindsor