Opentrons/opentrons #10127: Run App Fast Sim: Fast simulator treats 200uL filter tip like 300uL tips

alexjoel42, 2022-04-28:
# Overview

When using a 200uL filter tip (this doesn't happen for 10uL filter tips) the fast sim thinks the pipette can aspirate 300uL. E.G: if it is asked to aspirate 300uL it will just type out "aspirate 300uL" 

# Steps to reproduce
Make a protocol with 200uL tips and upload it onto the run app. 

## Example protocol 
``` python3

from opentrons import protocol_api
metadata = {'apiLevel': '2.11'}

def run(protocol: protocol_api.ProtocolContext):
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', 3) 
    stock = protocol.load_labware('opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical', 1)
    tubes_standard = protocol.load_labware('opentrons_24_tuberack_eppendorf_2ml_safelock_snapcap', 9) 
    tiprack200 = protocol.load_labware('opentrons_96_filtertiprack_200ul', 5) 
    tiprack200_2 = protocol.load_labware('opentrons_96_filtertiprack_200ul', 4)
    tiprack200_3 = protocol.load_labware('opentrons_96_filtertiprack_200ul', 2)
    p300 = protocol.load_instrument('p300_multi_gen2', mount='left', tip_racks=[tiprack200, tiprack200_2,tiprack200_3])
    ###WATER
    #Transfer Fe(II) working solution to eppendorph tubes
    p300.pick_up_tip()
    p300.aspirate(300, stock['A1'])
```
# Current behavior

1. Have a protocol where a pipette using a 200uL filter tip and has a command to aspirate 300uL 
2. Load protocol into opentrons_simulate, get ```opentrons.protocols.execution.errors.ExceptionInProtocolError: AssertionError [line 27]: Cannot aspirate more than pipette max volume```
3. Load the same protocol into the run app and it will say ```Aspirating 300.0 uL from A1 of Opentrons 10 Tube Rack with Falcon 4x50 mL, 6x15 mL Conical on 1 at 94.0 uL/sec`` and when the robot gets to the aspiration step, will error out.

[image]

**Note**
When using complex commands, the OT-2 will just split the volume as if it were 200uL while having the run log incorrect. 

# Expected behavior

Run app should through the same error as opentrons_simulate and opentrons_execute before protocol runs

amitlissack, 2022-04-29:
Is there a chance this is a dupe of #9503 ?

mcous, 2022-07-22:
Duplicate ticket, underlying issue fixed by #9550

closed 2022-07-22 by mcous