Opentrons/opentrons #10129: 5.1 Feedback:  Same protocol with different name seems to have same name in the app

y3rsh, 2022-04-29:
### Overview

I uploaded multiple protocols. In one instance, I uploaded one protocol with the name Johns Hopkins, and another Johns Hopkins (1), but in the app, they populate with the same name, so I can’t tell the difference.
[image]

### Current Behavior

Changing the file name does not reflect on upload.

### Expected Behavior

Changed file name does reflect on upload.

### Operating system

Mac

ecormany, 2022-04-29:
The current behavior is to only show the metadata `protocolName` if present and completely ignore filenames.

While more complicated, I think the most intuitive solution would be:
1. If `protocolName` is unique, show it only
2. If `protocolName` is not unique, show it followed by the filename (separated by en dash or parentheses)
3. If there is no `protocolName`, fall back to the filename

Of course, you could still import protocols with the same metadata name and filename — at that point distinguishing by timestamp seems sensible though.

nusrat813, 2022-04-29:
Expected behavior 

Eds feature request #2 - @mikecardarelli

closed 2022-04-29 by nusrat813

mikecardarelli, 2022-05-02:
New protocol naming ticket: https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/10175