Opentrons/opentrons #10131: 5.1 Feedback: Confusing display of multiple runs of a protocol

y3rsh, 2022-04-29:
### Overview

If I rerun multiple protocols of the same name they load in the app in a confusing order. I show a lot of duplicates, but I have to keep track of the timestamps. That is not ideal.

nusrat813, 2022-04-29:
@nusrat813 Sorting ticket open - what is the default sorting order? 

date of the last run is not in scope for 5.1.0

y3rsh, 2022-05-17:
@nusrat813 link sorting ticket.

nusrat813, 2022-05-18:
Sorting PR - https://github.com/Opentrons/opentrons/pull/10223

nusrat813, 2022-05-18:
The default sorting should be alphabetical

closed 2022-05-18 by nusrat813