Opentrons/opentrons #10150: Design QA: Device Details Page

mikecardarelli, 2022-04-29:


mmencarelli, 2022-05-11:
Merging in a few lines from #9851

Mini toast style

- [ ]  Add new mini toast style (view on figma)
- [ ]  Make sure the min-height on the mini toast card is 32px

mmencarelli, 2022-05-11:
Merging in more lines from #9851

Module cards

 Change the size of the img of the module to 60x54px
 Change the 'firmware update' toast in all of the module cards to be the new mini toast style
 Change the h/s 'module is too hot' toast in to be the new mini toast style
 Change the pipette 'POC missing' toast in to be the new mini toast style
 Change the pipette 'POC needed' toast in to be the new mini toast style
 Change the pipette 'POC recommended' toast in to be the new mini toast style

mmencarelli, 2022-06-03:
First round design qa #10611

mmencarelli, 2022-06-10:
#10730 first round design qa

koji, 2022-12-03:
Seems that this has been solved byt https://github.com/Opentrons/opentrons/pull/10732 and https://github.com/Opentrons/opentrons/pull/10771

closed 2022-12-03 by koji