Opentrons/opentrons #10204: TLC modal: link to support doc needs updating

mikecardarelli, 2022-05-04:


ecormany, 2022-05-04:
Realized we probably shouldn't be hunting for these manually. Here's a list of all the old-format URLs left in edge:
```
grep -rn "support.opentrons.com/en" app/src
app/src/pages/Robots/RobotSettings/CalibrationCard.tsx:52:  'https://support.opentrons.com/en/articles/3499692-how-calibration-works-on-the-ot-2'
app/src/pages/Devices/DevicesLanding/NewRobotSetupHelp.tsx:20:  'https://support.opentrons.com/en/collections/1559720-ot-2-get-started'
app/src/organisms/ProtocolSetup/RunSetupCard/ModuleSetup/MultipleModulesModal.tsx:31:  'https://support.opentrons.com/en/articles/5167312-using-modules-of-the-same-type-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/ModuleSetup/__tests__/MultipleModuleModal.test.tsx:43:      'https://support.opentrons.com/en/articles/5167312-using-modules-of-the-same-type-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/LabwareSetup/LabwareOffsetModal.tsx:29:  'https://support.opentrons.com/en/articles/3499692-how-positional-calibration-works-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/LabwareSetup/LabwareOffsetModal.tsx:31:  'http://support.opentrons.com/en/articles/5742955-how-labware-offsets-work-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/LabwareSetup/__tests__/LabwareOffsetModal.test.tsx:60:      'https://support.opentrons.com/en/articles/3499692-how-positional-calibration-works-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/LabwareSetup/__tests__/LabwareOffsetModal.test.tsx:70:      'http://support.opentrons.com/en/articles/5742955-how-labware-offsets-work-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/RobotCalibration/PipetteCalibration.tsx:32:  'https://support.opentrons.com/en/articles/3450143-gen2-pipette-compatibility'
app/src/organisms/ProtocolSetup/RunSetupCard/RobotCalibration/__tests__/DeckCalibrationModal.test.tsx:57:      'https://support.opentrons.com/en/articles/3499692-how-positional-calibration-works-on-the-ot-2'
app/src/organisms/ProtocolSetup/RunSetupCard/RobotCalibration/DeckCalibrationModal.tsx:30:  'https://support.opentrons.com/en/articles/3499692-how-positional-calibration-works-on-the-ot-2'
app/src/organisms/CalibrateTipLength/ConfirmRecalibrationModal.tsx:33:const CALIBRATION_URL = 'https://support.opentrons.com/en/articles/4523313'
app/src/organisms/ProtocolUpload/hooks/__tests__/RerunningProtocolModal.test.tsx:44:      'http://support.opentrons.com/en/articles/5742955-how-labware-offsets-work-on-the-ot-2'
app/src/organisms/ProtocolUpload/RerunningProtocolModal.tsx:34:  'http://support.opentrons.com/en/articles/5742955-how-labware-offsets-work-on-the-ot-2'
app/src/organisms/CalibrationPanels/NeedHelpLink.tsx:6:  'https://support.opentrons.com/en/collections/2426956-ot-2-calibration'
app/src/organisms/AppSettings/__tests__/ConnectRobotSlideout.test.tsx:98:      'https://support.opentrons.com/en/articles/2934336-manually-adding-a-robot-s-ip-address'
app/src/organisms/AppSettings/PreviousVersionModal.tsx:17:  'https://support.opentrons.com/en/articles/2393514-uninstall-the-opentrons-app'
app/src/organisms/AppSettings/GeneralSettings.tsx:45:  'https://support.opentrons.com/en/articles/1795303-get-started-update-your-ot-2#:~:text=It%E2%80%99s%20important%20to%20understand,that%20runs%20your%20protocols).'
app/src/organisms/AppSettings/ConnectRobotSlideout.tsx:28:  'https://support.opentrons.com/en/articles/2934336-manually-adding-a-robot-s-ip-address'
app/src/organisms/Devices/DevicesEmptyState.tsx:28:  'https://support.opentrons.com/en/collections/1559720-ot-2-get-started'
app/src/organisms/Devices/DevicesEmptyState.tsx:30:  'https://support.opentrons.com/en/articles/2687601-troubleshooting-connection-problems'
app/src/organisms/Devices/RobotSettings/RobotSettingsNetworking.tsx:37:  'https://support.opentrons.com/en/articles/2687586-get-started-connect-to-your-ot-2-over-usb'
app/src/organisms/Devices/RobotSettings/__tests__/RobotSettingsNetworking.test.tsx:195:      'https://support.opentrons.com/en/articles/2687586-get-started-connect-to-your-ot-2-over-usb'
app/src/organisms/Devices/ConnectionTroubleshootingModal.tsx:20:  'https://support.opentrons.com/en/articles/2687601-troubleshooting-connection-problems'
app/src/organisms/Devices/ProtocolRun/SetupPipetteCalibrationItem.tsx:36:  'https://support.opentrons.com/en/articles/3450143-gen2-pipette-compatibility'
```

They appear to all go to a login screen.

smb2268, 2022-05-17:
@aedocw @mikecardarelli Do we need to update all of the links in that comment?

mikecardarelli, 2022-05-20:
Updating this ticket to cover all the broken links in OT App

commit 19d370e3 (2022-05-20): fix(app): update out of date support links

fix #10204
PR #10388 (2022-05-20): fix(app): update out of date support links
fix #10204

# Overview

Change support links to new salesforce links

# Changelog

Updated all links according to this spreadsheet https://docs.google.com/spreadsheets/d/14QmcI6KgWcUN2m4MjVjKeDEkX-548biQUmJsGXMMgnc/edit#gid=0
Updated some of the link styling from this comment https://github.com/Opentrons/opentrons/pull/10364#pullrequestreview-980052547

# Review requests

Verify all of the links are correct as detailed in the spreadsheet
# Risk assessment

<!
commit fb4ed1f1 (2022-05-20): fix(app): update out of date support links (#10388)

fix #10204
closed 2022-05-20 by smb2268