Opentrons/opentrons #10211: 5.1 Feedback: [Design QA] Modal Dialogs: Delete this protocol #10167

mmencarelli, 2022-05-04:
### Overview

This feedback is r1 on the "delete this protocol" modal that appears when a user tries to delete a protocol on the T1 protocols page using the menu overflow button. 

### Expected Behavior

Link to reference frame on [Figma](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=11435%3A270225)

- [ ] (1) Update the modal to the new modal style. There's a lot of stuff in here, so I'll briefly outline below: 
- [ ] max-width of modal
- [ ] New icon for warning
- [ ] Make sure all text is the correct colour (dark-black)
- [ ] Make sure all buttons are the updated styling and we have all states included 
- [ ] (2) We're using the old overlay colour, so make sure to update that too! 

[image]

koji, 2022-05-17:
This has been solved by #10291

y3rsh, 2022-05-26:
I am not sure that the text is correct?
Should it say 
## Delete this protocol definition?
[image]

koji, 2022-06-08:
`Delete this protocol?` is the correct title for this modal.
This is confirmed by Mel
https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx?node-id=1297:90568#208168244