Opentrons/opentrons #10214: Add additional client-side restrictions on robot names to protect against robot-side bugs

SyntaxColoring, 2022-05-05:
# Overview

Currently, the Opentrons App validates robot names according to these rules: 

> You can use letters, numbers, spaces, special characters (!?$*) and emojis.
> 35 characters max

And in theory, the OT-2 can have a display name that's mostly arbitrary Unicode.

In practice, robot renaming has several bugs:

* https://github.com/Opentrons/opentrons/issues/9960
* https://github.com/Opentrons/opentrons/issues/10126
* https://github.com/Opentrons/opentrons/issues/10197
* https://github.com/Opentrons/opentrons/issues/10198

Since the OT-2 persists its display name across boots, hitting some of these bugs can have lasting effects. So, out of conservatism, in the initial release of the robot rename feature, we should limit input to names that are unlikely to trigger any of these bugs.

When these bugs have been resolved and we've had more time to vet our back-end code, we can lift these restrictions, and go back to the app's current restrictions of:

# Acceptance criteria

* The Opentrons App's robot rename field should only allow ASCII A-Z and ASCII 0-9. This means it should *disallow* spaces, special characters, and emoji. 
* The Opentrons App's robot rename field should continue to disallow names longer than 35 characters.
    * I believe the robot can safely handle ASCII names less than 63 characters long (and perhaps exactly 63—unclear). My understanding is that the current limit of 35 comes from the app's own design and testing concerns.
* Copy should also be updated.
* Since these new restrictions are intended to be temporary, I suggest leaving a `# TODO` comment pointing back to this ticket for traceability.

koji, 2022-05-06:
@SyntaxColoring 
We changed the rules of a robot name

>Choose a name for your robot. You can use letters, numbers, spaces, and these special characters: !?$*’-_.
We disallow a user to use space for the first letter.

SyntaxColoring, 2022-05-06:
Noted, thanks. I think this ticket still stands. Anything not ASCII A-Z 0-9 is at risk of not being handled correctly.

koji, 2022-05-17:
> @SyntaxColoring Are lowercase letters acceptable?

@mikecardarelli  I got confirmation from Max. The lowercase letters are acceptable.

PR #10318 (2022-05-17): fix(app): Update robot rename rule

# Overview
This will fix #10214

The new rule is that allows users to use a-z, A-Z and 0-9.

regex
`/^([a-zA-Z0-9]{0,35})$/`
https://regexr.com/

[image]

# Changelog
- Updated RenameRobotSlideout regex
- Updated the test for a new regex pattern
- Cleaned up `device_settings` 

# Review requests

# Risk assessment
low
closed 2022-05-17 by koji