Opentrons/opentrons #10246: 5.1 Feedback: Toggling the Light from laptop 1 does not reflect on laptop 2

emilywools, 2022-05-09:
### Overview

2 people's laptops are on the network with robot A.

Laptop 1 toggles the light on robot A. On laptop 2, the robot light status does not change.

### Current Behavior

the robot's light status does not change on a 2nd comptuter.

### Expected Behavior

the robot's light status should be reflected on any laptop thats on the network and on that robot's detail page.

### Operating system

Mac

nusrat813, 2022-05-10:
I believe this is present in 5.0.0 as well
Can downgrade and check

nusrat813, 2022-05-26:
Downgraded to 5.0.2 and the light toggle status is working, so this is def a bug

mcous, 2022-05-26:
I think this might be another issue similar to the module stuff. The robot settings page only asks the server for the lights status when the robot settings page is first rendered. It should probably be moved to `react-query` and with a slow default stale (poll) interval.

On my machine, the lights status updates correctly if I navigate away from the robot and come back to it

commit 5015eb6a (2022-06-08): refactor(app): poll robot lights status (#10645)

poll robot lights status on a 5 second interval within the RobotOverview component

closes #10246