Opentrons/opentrons #10258: bug: heater shaker modules endpoint not matching engine status

shlokamin, 2022-05-10:
# Overview

When setting a temperature/shake speed on a heater shaker, the engine successfully executes the given command, but the module state from the modules endpoint does not update accordingly.

# Steps to reproduce
In the heater shaker slideout, set a temperature of 40 degrees
If you inspect your network tab, you should see the command was successfully queued (and gets executed immediately)
The module data coming from the modules endpoint, however, never updates. The `currentTemperature` and `temperatureStatus` fields never get updated.

The same behavior happens when setting a shake speed — the engine successfully executes the command, but the modules endpoint never actually updates with new state.

sanni-t, 2022-06-10:
Was this tested on a dev server with simulated module or on a robot with a real heater-shaker connected?

shlokamin, 2022-06-10:
real heater shaker!

sanni-t, 2022-06-10:
> real robot + heater shaker!

Cool, that makes sense. So from my investigation, this isn't an endpoints issue. The endpoints update fine when using a dev server. There are bugs in the robot's communication with the heater-shaker which puts us into a situation where either the heater-shaker actually doesn't perform the communicated action or the robot is unable to understand the response from the heater-shaker. So far these are/were #10605, #10707, [#318](https://github.com/Opentrons/opentrons-modules/issues/318), #10249.

Will close this ticket out once it's verified that fixing the above bugs fixes all module control commands issues

PR #10727 (2022-06-13): fix(api): fix multiple h/s bugs
Closes #10605, Closes #10258

# Overview

Has fixes for:
- h/s not deactivating via `heaterShaker/deactivateHeater` command
- temperature status parsing error

# Review requests

Test the changes with a real h/s with firmware version v0.4.4+. Push changes to a robot, connect the h/s to the robot and check that all h/s **HTTP** commands succeed and the module's status response reflects its updated state. Note that you'll need to wait until a second after the command is fully executed in order for the status to update in the backend.

I've tested this on my robot  [... cut]
commit b8c7c887 (2022-06-15): fix(api): fix multiple h/s bugs (#10727)

Closes #10605, Closes #10258
closed 2022-06-13 by mcous