Opentrons/opentrons #10263: 5.1 Feedback:  [Design QA] Robot Settings Page#10151 & Robot Settings > Advanced > Usage settings styling is off #10106

mmencarelli, 2022-05-10:
### Overview

Round 1 design QA for Robot Settings page

### Current Behavior

View the [annotated Figma frame](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=11756%3A288136)

----
**Calibration**
- [ ] (1) Update the font styling here to the updated label font size  (11 font size, 12px line height)
- [ ] (2) Update font colour to our 'dark-black'
- [ ] (3) Update this link to our 'link/primary/blue/internal' styling. There's a heavier font weight and a hover state
- [ ] (4) Update the tertiary button styling. Need correct hover/pressed/potentially disabled state (cant activate disabled using devtools)
- [ ] (5) Are we nixing the rest of the content on this page? If so, let's nix that last line beneath content. Also, let me know so I can update designs. 

**Networking**
- [ ] (1) Update all text to our 'dark-black' on this page
- [ ] (2) Should be semi-bold
- [ ] (3) Update to our 'input/field-with-label' component. The label should be our label styling, currently our p styling
- [ ] (4) Missing 24px margin
- [ ] (5) Double the size of the icon, should be 12x12
- [ ] (6) Missing hover state on the primary link styling
- [ ] (7) 1rem spacing between link and "looking for
- [ ] (8) Need to update these icons

**Advanced**
- [ ] (1) Bit too much spacing here, should be 1rem from the edge of the container
- [ ] (2) Update all text to our 'dark-black' on this page
- [ ] (3) Should be 1 rem beneath each section header (about, usage settings, trouble shooting, legacy settings). Also, making a last-minute decision to update the headers from h3's to h2's and semibold to help add some distinction.
- [ ] (4) Update tertiary button style
- [ ] (5) Needs our link styling (semibold, hover states)
- [ ] (6) Missing the 'update now' button? This displays when the toast saying there's an update available. 
- [ ] (7) Upping the margin to 40px between each section (sorry, this is new!)
- [ ] (8) Update these to our toggle component
- [ ] (9) Update size of font, size of icon and add the states in
- [ ] (10) Looks like a margin-bottom of 1rem snuck in there

[image]
[image]
[image]
[image]

koji, 2022-05-12:
https://user-images.githubusercontent.com/474225/168160607-9fd4c41c-f880-4a47-bd0b-a7ef239b071c.mov

koji, 2022-05-12:
[image]

koji, 2022-05-12:
@mmencarelli 
[image]

mmencarelli, 2022-05-12:
> @mmencarelli [image]

Yup, I can see the confusion. If you step out two levels you'll see that the component includes some padding around the vector. So vector is 8x8, but there should be some padding around it that cause it to be 12x12. [Video here](https://www.loom.com/share/35a60164e3344afa9ab61fa54847006b)

koji, 2022-05-13:
@mmencarelli thank you for the video.

After checking components, noticed that ExternalLink's line-height was 16px instead of 20px.

The following is line-height 20px. What do you think of these?

0.5 rem 8px
[image]

0.6 rem 9.6px
[image]

0.75 rem 12px
[image]

koji, 2022-05-13:
Default fontColor `#4a4a4a`

koji, 2022-05-13:
[image]

mmencarelli, 2022-05-16:
> @mmencarelli thank you for the video.
> 
> After checking components, noticed that ExternalLink's `line-height` was 16px instead of 20px and added `align-items: center` to centralize text and icon vertically.
> 
> The following is `line-height` 20px. What do you think of these?
> 
> 0.5 rem 8px [image]
> 
> 0.6 rem 9.6px [image]
> 
> 0.75 rem 12px [image]

Love the options! 0.5 rem 8px is the one :)

koji, 2022-05-16:
@mmencarelli Thank you for your feedback!

koji, 2022-05-16:
- [ ] icon size 0.5 rem for ExternalLink

PR #10290 (2022-05-17): fix(app): App fix robot settings design feedback

# Overview
This will fix #10263 
Reflected the feedback on `RobotSettings`.

# Changelog
- Updated RobotSettingsCalibration styling
- Updated RobotSettingsNetworking styling
- Updated RobotSettingsAdvanced styling
- Update linkPsemiBold line-height
- Change `GlobalStyle` color to `darkBlack`

# Review requests

# Risk assessment
low
closed 2022-05-17 by y3rsh