Opentrons/opentrons #10279: 6.0 Feedback: robot ignoring `delay()` commands in Python protocols

ecormany, 2022-05-12:
### Overview

On v5.1.0-beta.0 of the robot server, all `delay()` commands are ignored. Downgrading to 5.0.2 resolves the issue.

### Current Behavior

When running a Python protocol on a robot running v5.1.0-beta.0 of the robot server, `delay()` commands have no effect. They appear in the run log for the protocol, but all other actions are performed as quickly as possible, with no delays.

### Expected Behavior

Each `delay()` action should pause the robot's movements for exactly as long as specified and as shown in the run log.

### Steps To Reproduce

1. Import a protocol that includes `delay()` commands.
[move.py.zip](https://github.com/Opentrons/opentrons/files/8682792/move.py.zip)

2. Run the protocol on a robot running the beta server.

Downgrading to robot server 5.0.2 restores the expected behavior. Upgrading to the beta again reintroduces the bug.

### Operating system

Mac

### Robot setup or anything else?

Robot Server Version
v5.1.0-beta.0

Robot Serial Number
OT2CEP20210812R11

Firmware Version
v1.1.0-25e5cea

y3rsh, 2022-05-17:
@SyntaxColoring I reproduced this from edge.

SyntaxColoring, 2022-05-18:
Also reproduced. Looking into this now.

closed 2022-05-18 by SyntaxColoring

y3rsh, 2022-05-19:
Delays respected upon retesting.

PR #10334 (2022-05-18): fix(api): Fix delays returning immediately
# Overview

Fixes #10279.

# Explanation

This bug was a missing `await` that got hidden in a sneaky way.

Here, we have an async method that returns an async task:

https://github.com/Opentrons/opentrons/blob/b1fd19f6fa7cb2dc8cab2cac43e0015d07b561b1/api/src/opentrons/hardware_control/execution_manager.py#L53-L56

And here, we try to call that method and then wait for the returned task to finish:

https://github.com/Opentrons/opentrons/blob/b1fd19f6fa7cb2dc8cab2cac43e0015d07b561b1/api/src/opentrons/hardware_control/execution_manager.py#L108

But notic [... cut]