Opentrons/opentrons #10289: 6.0 Feedback:  [Design QA] Global Components, App size + min-width #10147

mmencarelli, 2022-05-13:
### Overview

This is some some overall feedback on viewport sizing. 

### Current Behavior

- [ ] Currently the app opens up at 1020x768px but I'd like to change it to 800x760px to match the designs in Figma
- [ ] Can we add a min-width to the app of 600px

b-cooper, 2022-05-13:
Let's try to get this one in before the first release candidate is tagged

ecormany, 2022-05-13:
Min width (and height) would also address https://github.com/Opentrons/opentrons/issues/9651.

closed 2022-05-24 by vabruzzo

y3rsh, 2022-05-26:
default size is 800X760 and min width is 600
this is working on mac,linux, and windows.