Opentrons/opentrons #10300: 6.0 Feedback:  Can't select options for Pipettes and Modules Section under Devices Tab

braldrich, 2022-05-16:
### Overview

In the devices section of the app, if I can't select anything from the pipettes' drop down menu. Noticed this specifically when I tried to recalibrate pipette offset from this menu, but noticed I also can't access "detach pipette" or "View pipette settings." I click on them and there is no response. 

Tested magnetic module menu just to see and it works fine, able to set engage height and access "about module" page. 

[image]

### Current Behavior

Currently I cannot recalibrate pipette offset, detach pipette, or view pipette settings from this menu for either the left or right mounted pipette. 

### Expected Behavior

The menu should work, these options should take me to the appropriate action. 

### Steps To Reproduce

See screenshot in overview, go to the pipettes and modules section within the devices section, click on the ellipses, and try selecting any of the three options. This has persisted through a downgrade of the app and my robot to 5.0.2 and upgrading again. 

### Operating system

Mac

### Robot setup or anything else?

I am connected to my robot via USB, I have a P300 single and multi mounted. This issue

ecormany, 2022-05-16:
Already fixed by #10009

closed 2022-05-16 by ecormany

nusrat813, 2022-06-08:
Verified Fixed