Opentrons/opentrons #10329: 6.0 Feedback:  Can we change how we treat Labware Position Check as optional?

alexjoel42, 2022-05-18:
### Overview

Declaring that Labware Position Check is optional doesn't match the customer experiences of those that have reached out to me or my own experience. Can we be looking for whether this is truly optional in the beta testing? 

If it is really optional I think that customers will ignore it after seeing negligible Labware Offsets. 

Is this something we can track through app data for 5.x.x? 

That being said, totally understand removing the "optional flag" isn't something the team would be interested in this round.  

### Current Behavior

We put it as optional and customers come in with the expectation that it is optional and it is hard to explain from Technical Support's perspective why "it is optional for some folks and not for you"

### Expected Behavior

In practice, customers should be doing Labware Position Check for every new protocol unless they perform the labware position check and find that the Labware Offsets are negligible (like our team would expect them to be) 

### Operating system

Mac

### Robot setup or anything else?

Wi-Fi

emilywools, 2022-05-19:
hey @alexjoel42 thank you for sharing this feedback. Are you requesting that we change the logic of LPC such that performing it is required to proceed to run, or just removing the "optional" label from it?

[image]

alexjoel42, 2022-05-19:
Hello @emilywools!

## Recommending preference
I think putting it as recommended would be better than optional. Putting what I see as the two "worst" situations for optional/recommended

1. Optional: The customer's robot crashes because they didn't do the labware position check (which we have seen happen in Support). They reach out to Support and we identify the source of the crash after troubleshooting. 
2. Recommended: The customer continuously calibrates with Labware Offsets as zero. People that are frustrated reach out to Support and we tell them they can just not do it. 

## Making it mandatory - Alex is unsure

Whether it's necessary or not is a call I would be uncomfortable making. I am biased toward making it necessary for any unsaved protocols, but that's because the only people I talk to about it are customers that have calibration issues.

 If most folks external/internally need the labware position check before every run (otherwise the run doesn't work), even after they pass calibration health checks, I would feel more comfortable.

emilywools, 2022-05-19:
for now, let's update the copy from "optional" to "recommended." 

Here's the ticket: https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/10359

closed 2022-05-19 by emilywools