Opentrons/opentrons #10360: bug: pillow dependency in /hardware fails to install on arm64

y3rsh, 2022-05-19:
# Overview

`make -C hardware setup` fails on arm64 machine.
computer setup per [DEV_SETUP.md](https://github.com/Opentrons/opentrons/blob/edge/DEV_SETUP.md)

```
make -C hardware setup
make[2]: Entering directory '/Users/johnmcvey/github/opentrons/opentrons/hardware'
python -m pipenv sync --dev --python 3.7
Virtualenv already exists!
Removing existing virtualenv...
Creating a virtualenv for this project...
Pipfile: /Users/johnmcvey/github/opentrons/opentrons/hardware/Pipfile
Using /Users/johnmcvey/.pyenv/versions/3.7.13/bin/python3.7m (3.7.13) to create virtualenv...
⠸ Creating virtual environment...created virtual environment CPython3.7.13.final.0-64 in 224ms
  creator CPython3Posix(dest=/Users/johnmcvey/.local/share/virtualenvs/hardware-PicR8xfW, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/Users/johnmcvey/Library/Application Support/virtualenv)
    added seed packages: pip==22.0.4, setuptools==62.1.0, wheel==0.37.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator

✔ Successfully created virtual environment!
Virtualenv location: /Users/johnmcvey/.local/share/virtualenvs/hardware-PicR8xfW
Installing dependencies from Pipfile.lock (9ab443)...
An error occurred while installing pillow==9.1.0; python_version >= '3.7' --hash=sha256:eb1b89b11256b5b6cad5e7593f9061ac4624f7651f7a8eb4dfa37caa1dfaa4d0 --hash=sha256:463acf531f5d0925ca55904fa668bb3461c3ef6bc779e1d6d8a488092bdee378 --hash=sha256:5a3ecc026ea0e14d0ad7cd990ea7f48bfcb3eb4271034657dc9d06933c6629a7 --hash=sha256:c7110ec1701b0bf8df569a7592a196c9d07c764a0a74f65471ea56816f10e2c8 --hash=sha256:17869489de2fce6c36690a0c721bd3db176194af5f39249c1ac56d0bb0fcc512 --hash=sha256:c870193cce4b76713a2b29be5d8327c8ccbe0d4a49bc22968aa1e680930f5581 --hash=sha256:01ce45deec9df310cbbee11104bae1a2a43308dd9c317f99235b6d3080ddd66e --hash=sha256:5cfca31ab4c13552a0f354c87fbd7f162a4fafd25e6b521bba93a57fe6a3700a --hash=sha256:f401ed2bbb155e1ade150ccc63db1a4f6c1909d3d378f7d1235a44e90d75fb97 --hash=sha256:8782189c796eff29dbb37dd87afa4ad4d40fc90b2742704f94812851b725964b --hash=sha256:c5b0ff59785d93b3437c3703e3c64c178aabada51dea2a7f2c5eccf1bcf565a3 --hash=sha256:de344bcf6e2463bb25179d74d6e7989e375f906bcec8cb86edb8b12acbc7dfef --hash=sha256:af79d3fde1fc2e33561166d62e3b63f0cc3e47b5a3a2e5fea40d4917754734ea --hash=sha256:55dd1cf09a1fd7c7b78425967aacae9b0d70125f7d3ab973fadc7b5abc3de652 --hash=sha256:3f42364485bfdab19c1373b5cd62f7c5ab7cc052e19644862ec8f15bb8af289e --hash=sha256:69e5ddc609230d4408277af135c5b5c8fe7a54b2bdb8ad7c5100b86b3aab04c6 --hash=sha256:3fddcdb619ba04491e8f771636583a7cc5a5051cd193ff1aa1ee8616d2a692c5 --hash=sha256:7a053bd4d65a3294b153bdd7724dce864a1d548416a5ef61f6d03bf149205160 --hash=sha256:255c9d69754a4c90b0ee484967fc8818c7ff8311c6dddcc43a4340e10cd1636a --hash=sha256:6b6d4050b208c8ff886fd3db6690bf04f9a48749d78b41b7a5bf24c236ab0165 --hash=sha256:21dee8466b42912335151d24c1665fcf44dc2ee47e021d233a40c3ca5adae59c --hash=sha256:25023a6209a4d7c42154073144608c9a71d3512b648a2f5d4465182cb93d3477 --hash=sha256:c9efef876c21788366ea1f50ecb39d5d6f65febe25ad1d4c0b8dff98843ac244 --hash=sha256:a336a4f74baf67e26f3acc4d61c913e378e931817cd1e2ef4dfb79d3e051b481 --hash=sha256:66822d01e82506a19407d1afc104c3fcea3b81d5eb11485e593ad6b8492f995a --hash=sha256:4fe29a070de394e449fd88ebe1624d1e2d7ddeed4c12e0b31624561b58948d9a --hash=sha256:a598d8830f6ef5501002ae85c7dbfcd9c27cc4efc02a1989369303ba85573e58 --hash=sha256:97bda660702a856c2c9e12ec26fc6d187631ddfd896ff685814ab21ef0597033 --hash=sha256:0c51cb9edac8a5abd069fd0758ac0a8bfe52c261ee0e330f363548aca6893595 --hash=sha256:a325ac71914c5c043fa50441b36606e64a10cd262de12f7a179620f579752ff8 --hash=sha256:a5eaf3b42df2bcda61c53a742ee2c6e63f777d0e085bbc6b2ab7ed57deb13db7 --hash=sha256:35be4a9f65441d9982240e6966c1eaa1c654c4e5e931eaf580130409e31804d4 --hash=sha256:82283af99c1c3a5ba1da44c67296d5aad19f11c535b551a5ae55328a317ce331 --hash=sha256:aea7ce61328e15943d7b9eaca87e81f7c62ff90f669116f857262e9da4057ba3 --hash=sha256:fb89397013cf302f282f0fc998bb7abf11d49dcff72c8ecb320f76ea6e2c5717 --hash=sha256:c24f718f9dd73bb2b31a6201e6db5ea4a61fdd1d1c200f43ee585fc6dcd21b34 --hash=sha256:ed742214068efa95e9844c2d9129e209ed63f61baa4d54dbf4cf8b5e2d30ccf2 --hash=sha256:8d79c6f468215d1a8415aa53d9868a6b40c4682165b8cb62a221b1baa47db458! Will try again.
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 49/49 — 00:03:02
Installing initially failed dependencies...
[InstallError]:   File "/Users/johnmcvey/.pyenv/versions/3.7.13/lib/python3.7/site-packages/pipenv/cli/command.py", line 696, in sync
[InstallError]:       system=state.system
[InstallError]:   File "/Users/johnmcvey/.pyenv/versions/3.7.13/lib/python3.7/site-packages/pipenv/core.py", line 2892, in do_sync
[InstallError]:       system=system,
[InstallError]:   File "/Users/johnmcvey/.pyenv/versions/3.7.13/lib/python3.7/site-packages/pipenv/core.py", line 1312, in do_init
[InstallError]:       pypi_mirror=pypi_mirror,
[InstallError]:   File "/Users/johnmcvey/.pyenv/versions/3.7.13/lib/python3.7/site-packages/pipenv/core.py", line 900, in do_install_dependencies
[InstallError]:       retry_list, procs, failed_deps_queue, requirements_dir, **install_kwargs
[InstallError]:   File "/Users/johnmcvey/.pyenv/versions/3.7.13/lib/python3.7/site-packages/pipenv/core.py", line 796, in batch_install
[InstallError]:       _cleanup_procs(procs, failed_deps_queue, retry=retry)
[InstallError]:   File "/Users/johnmcvey/.pyenv/versions/3.7.13/lib/python3.7/site-packages/pipenv/core.py", line 703, in _cleanup_procs
[InstallError]:       raise exceptions.InstallError(c.dep.name, extra=err_lines)
[pipenv.exceptions.InstallError]: Collecting pillow==9.1.0
[pipenv.exceptions.InstallError]:   Using cached Pillow-9.1.0.tar.gz (49.8 MB)
[pipenv.exceptions.InstallError]:   Preparing metadata (setup.py): started
[pipenv.exceptions.InstallError]:   Preparing metadata (setup.py): finished with status 'done'
[pipenv.exceptions.InstallError]: Building wheels for collected packages: pillow
[pipenv.exceptions.InstallError]:   Building wheel for pillow (setup.py): started
[pipenv.exceptions.InstallError]:   Building wheel for pillow (setup.py): finished with status 'error'
[pipenv.exceptions.InstallError]:   error: subprocess-exited-with-error
[pipenv.exceptions.InstallError]:
[pipenv.exceptions.InstallError]:   × python setup.py bdist_wheel did not run successfully.
[pipenv.exceptions.InstallError]:   │ exit code: 1
[pipenv.exceptions.InstallError]:   ╰─> [188 lines of output]
[pipenv.exceptions.InstallError]:       running bdist_wheel
[pipenv.exceptions.InstallError]:       running build
[pipenv.exceptions.InstallError]:       running build_py
[pipenv.exceptions.InstallError]:       creating build
[pipenv.exceptions.InstallError]:       creating build/lib.macosx-12.2-arm64-cpython-37
[pipenv.exceptions.InstallError]:       creating build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/MpoImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/ImageMode.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/PngImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/XbmImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/PcxImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/SunImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/ImageFile.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/SpiderImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/TarIO.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/FitsStubImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/MpegImagePlugin.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[pipenv.exceptions.InstallError]:       copying src/PIL/BdfFontFile.py -> build/lib.macosx-12.2-arm64-cpython-37/PIL
[... cut ...]
[pipenv.exceptions.InstallError]: hint: See above for output from the failure.
ERROR: Couldn't install package: pillow
 Package installation failed...
  ☤  ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/1 — 00:00:05
make[2]: *** [Makefile:55: setup] Error 1
make[2]: Leaving directory '/Users/johnmcvey/github/opentrons/opentrons/hardware'
make[1]: *** [Makefile:65: hardware-py-setup] Error 2
make[1]: Leaving directory '/Users/johnmcvey/github/opentrons/opentrons'
make: *** [Makefile:61: setup-py] Error 2
```

ecormany, 2022-05-19:
worked for me on `edge`. i have `jpeg` (the missing dependency) installed via homebrew.

> [pipenv.exceptions.InstallError]:       The headers or library files could not be found for jpeg,
> [pipenv.exceptions.InstallError]:       a required dependency when compiling Pillow from source.

y3rsh, 2022-05-20:
Thanks @ecormany I could have at least glanced 🤣

PR #10375 (2022-05-24): docs(dev_setup): add jpeg install instructions for macOS arm fixing pillow install
# Overview

closes #10360

# Risk assessment
None, dev setup documentation only.
commit 360d45e3 (2022-05-24): docs(dev_setup): add jpeg install instructions for macOS arm fixing pillow install (#10375)

* docs(dev_setup): add jpeg install to macOS arm instructions to allow pillow to install

closes #10360
closed 2022-05-24 by y3rsh