Opentrons/opentrons #10384: 6.0 Feedback:  "See how to setup a new robot" link on device landing page is broken

reachojhaneha, 2022-05-20:
### Overview

"See how to setup a new robot" link on device landing page is broken

### Current Behavior

[image]

When i click on "see how to setup a new robot" link is broken, there is no response

### Expected Behavior

upon clicking the link, a modal should opeen

### Steps To Reproduce

Go to device landing page -> click on "See how to setup a new robot"

koji, 2022-05-20:
@reachojhaneha Does this happen to you when you click the link right after starting the app?

reachojhaneha, 2022-05-20:
@koji , this happens all the time

koji, 2022-05-20:
@reachojhaneha could you try this branch `app_fix-see-how-to-setup-new-robot-modal`?

closed 2022-05-23 by koji

y3rsh, 2022-05-26:
[image]

PR #10390 (2022-05-23): fix(app): fix See how to setup a new robot link

# Overview
This fix #10384

https://user-images.githubusercontent.com/474225/169609617-b5cce62f-c421-4570-9412-c8f9fc100c76.mov

# Changelog
- Added `level` to `Portal` 

# Review requests
- The modal shows up right after starting up which means Click Device and click `See how to set up a new robot`

# Risk assessment
low