Opentrons/opentrons #10424: feat: labware creator test guide copy fixes

skyewindsor, 2022-05-24:
# Overview

There are a couple of copy fixes that need to be made to the Labware Creator labware definition test guide. Test guide can be found here and is linked at the bottom of the [Labware Creator](https://labware.opentrons.com/create/) page: https://opentrons-publications.s3.us-east-2.amazonaws.com/labwareDefinition_testGuide.pdf

# Implementation details

At step 1c, a line of copy needs to be added to say: "If your labware is a multi-well reservoir, the test will skip to Step 2b."
[image]

At the bottom of the test guide, this sentence needs to be fixed to say "If your robot still misses the calibration points on the deck when you run this protocol contact support@opentrons.com". Make sure "teh" is corrected to "the" and there is a period in the support email address instead of a comma. 
[image]

# Acceptance criteria

- [ ] Step 1c has the new line of copy in it
- [ ] The typos at the end of the test guide are fixed

skyewindsor, 2022-06-01:
Hey @chazmanc I’m using ZenHub in GitHub, [click this link](https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/board) to join my workspace and see other features available in GitHub or [download the ZenHub extension](https://www.zenhub.com/extension) and sign up with your GitHub account. 
  _Posted using [ZenHub](https://www.zenhub.com)_

mikecardarelli, 2022-09-09:
New JIRA ticket: https://opentrons.atlassian.net/browse/RAUT-198

closed 2022-09-09 by mikecardarelli