Opentrons/opentrons #10431: 6.0 Feedback: Robot update modal closes the first time you try to use it

mcous, 2022-05-24:
### Overview

The first time I try to use the robot update modal, it closes without doing anything.

### Current Behavior

[image]

### Expected Behavior

The robot update wizard should proceed when I click "View Robot Update"

### Steps To Reproduce

1. Using the production v5.0.2 app, get an OT-2 onto v5.0.2
2. Open the development app from `edge`
3. Navigate to "Devices"
4. Click on "View update" > "Update robot now" > "View robot update"
5. Wizard closes instead of proceeding

If you then go through steps 4 and 5 again, the robot update will proceed normally.

### Operating system

Mac

### Robot setup or anything else?

- My OT-2R was on v5.0.2 production
- My app was on `edge` (a731395)
    - As of this commit, the app is unable to update robots due to a bug in `fetchModulesEpic`
    - However, the premature wizard close seems orthogonal
- Feel free to close this ticket in favor of #9808 and/or #9809

nusrat813, 2022-05-26:
Retest on alpha

y3rsh, 2022-05-26:
https://user-images.githubusercontent.com/502770/170560462-f2c0bf3a-feb8-4e97-add9-125a2ceef5e8.mp4

closed 2022-06-06 by jerader

y3rsh, 2022-06-07:
verified fixed

PR #10592 (2022-06-06): fix(app): robot update Modal CTA fix
closes #10431

# Overview

When trying to update the robot through the banner in the robot card, it won't properly and will instead redirect you to the robot overview page. This PR fixes that bug.

# Changelog

- add prop to `updateRobotBanner` so opening the `updateBuildroot` modal happens outside of the `onClick` in the flex in `RobotCard`. The `onClick` was causing the issue

- fix all instances of `updateRobotBanner`
- move `UpdateBuildroot` folder to `organisms/RobotSettings` and fix affected imports and tests (same thing that will be done in this PR:  [... cut]
commit 33fd4abd (2022-06-06): fix(app): robot update Modal CTA fix (#10592)

closes #10431