Opentrons/opentrons #10433: 6.0 Feedback:  white screen after renaming the robot name

reachojhaneha, 2022-05-24:
### Overview

change the name of a robot -> then click on the advance tab on robot-settings. App goes blank

### Current Behavior

[image]

[image]

### Steps To Reproduce

1. Goto robot-settings
2.  Goto to advance tab -> rename the robot -> enter a valid name -> click on rename robot button
3.  then click on advance tab -> app goes blank

### Operating system

Mac

koji, 2022-05-24:
@reachojhaneha  Did you use a real robot when you rename the robot name?

reachojhaneha, 2022-05-25:
@koji thats correct i used my real robot for renaming

koji, 2022-05-25:
@reachojhaneha thanks.
I haven't seen that case before. I will go to the office tomorrow and try to reproduce that.

PR #10463 (2022-05-26): fix(app): fix white screen issue after renaming clicking advanced tab

# Overview
This will fix #10433
The white screen happened because of an unexpected robot type(unreachable) and the Troubleshooting component breaks the entire app. This unexpected robot type issue would be caused by react tree rendering issue.
This PR changed the redirect path from `robot-settings` page to `robot details` page to avoid the Troubleshooting component button disabled issue. This rendering issue will be solve by another PR.

# Changelog
- Update Troubleshooting components (previously passed `robot` to the compo [... cut]
closed 2022-05-26 by koji