Opentrons/opentrons #10434: 6.0 Feedback: app update available modal sized incorrectly

mcous, 2022-05-24:
### Overview

The size and overlay of the app's "App update available" modal seems wrong.

### Current Behavior

[image]

### Expected Behavior

Going off v5, I'd expect to see:

- A darker opacity on the overlay, so it doesn't look like the nav UI is interactable
- A consistent margin around the entire modal

### Steps To Reproduce

1. Set your app's channel to `stable`
2. Close and relaunch the dev app from `edge`

### Operating system

Mac

nusrat813, 2022-05-26:
Retest on alpha

jerader, 2022-06-07:
@emilywools, what opacity do we want the background to be? Right now, it is at 35%

mcous, 2022-06-10:
The vertical spacing still looks wrong to me, why was this closed?

## v5.x (expected vertical spacing)

[image]

## v6.0.0-alpha.2

[image]

mcous, 2022-06-30:
Still present in `release_6.0.0` on 5d775be

[image]

PR #10635 (2022-06-07): refactor(app): adjust update app modal position and background opacity
closes #10434

# Overview

As outlined in the ticket, the update app modal launched during start up and when you navigate to the App general settings page is located in different positions. This PR fixes it as well as darkens the background

# Changelog

- edit `UpdateAppModal` background color
- add `Portal` to `GeneralSettings`

# Review requests

When you have a app update available:
- freshly launch the app, the modal should be centered in the middle of the overall app dimensions.
- navigate to the App general settings, [... cut]
commit b96c04a1 (2022-06-07): refactor(app): adjust update app modal position and background opacity (#10635)

closes #10434
closed 2022-07-01 by b-cooper