Opentrons/opentrons #10435: 6.0 Feedback: cannot re-install current software version to robot

mcous, 2022-05-24:
### Overview

In the v5 release of the app, one can "re-install" the current version of the robot software. This is an important feature that can get around some weird issues, especially those caused by meddling with the robot's read-only filesystem.

### Current Behavior

There is no re-install button in the robot's advanced settings:

[image]

### Expected Behavior

There should be a re-install button, according to the designs:

[image]

### Steps To Reproduce

1. Open the app
2. Navigate to a robot's advanced settings page

### Operating system

Mac

### Robot setup or anything else?

Tested on `edge` at 04f24cd

mcous, 2022-06-10:
Still missing issue in `alpha.2`

## v6.0.0-alpha.2

[image]

## v5.x (expected)

[image]

[image]

bndo, 2022-06-23:
Closed by #10773

closed 2022-06-23 by bndo

y3rsh, 2022-06-30:
successfully re-installed 6.0.0-alpha.3 on my robot from the app.

commit ace769c6 (2022-06-17): feat(app): add reinstall button to robot advanced settings (#10773)

close #10435