Opentrons/opentrons #10476: 6.0 Feedback:  Experience > Getting to deck calibration with an uncalibrated robot is not obvious

y3rsh, 2022-05-26:
### Overview

[image]

### Expected Behavior

Could the alerts launch deck calibration?
Could there be an alert at top to launch deck calibration? 

### Operating system

Mac

nusrat813, 2022-06-01:
@emilywools thoughts on this

emilywools, 2022-06-01:
you're right that this is not a good experience. Morgan just wrapped up a project to make robot calibration much more clear, but its not in scope for 6.0

nusrat813, 2022-06-01:
In scope for 6.2.0 - https://www.figma.com/file/8R6jZdYMMzAh6aeRkzHz4l/New-user-calibration?node-id=1598%3A114042

closed 2025-12-15 by sfoster1