Opentrons/opentrons #10477: 6.0 Feedback:  DB reset not integrated into factory reset in the app

y3rsh, 2022-05-26:
### Overview

If I check everything and factory reset, the DB was not deleted.
[image]
[image]

### Current Behavior

I believe the capability was just not yet integrated.

### Operating system

Mac

koji, 2022-05-26:
@y3rsh 

Could you tell me your test environment since actually, the reset function is working on my end?

y3rsh, 2022-05-31:
@koji 
The data like calibrations is all reset as expected.
This is specifically about the newly persisted in 6.0 data.
The robot now persists protocols, runs, and analyses through reboot.
The clear out that data, Robot Services added #10395 so that the DB is cleared out.

koji, 2022-05-31:
@y3rsh you mean only runsHistory reset doesn't work? 
If so, the issue will be solved by #10461

koji, 2022-06-02:
Closed by  #10461

closed 2022-06-02 by koji

nusrat813, 2022-06-07:
Verified fix