Opentrons/opentrons #10483: 6.0 Feedback:  App does not display spinner when refresh button in devices tab clicked

nusrat813, 2022-05-26:
### Current Behavior

The app does not display a spinner when refresh button to find robots is clicked 

### Expected Behavior

[image]

### Steps To Reproduce

1. Click refresh button
[image]
 

2. See no spinner is displayed, blank app 
[image]
 

### Operating system

Mac

nusrat813, 2022-06-01:
@b-cooper - PR #10399

y3rsh, 2022-06-07:
refresh button works but a couple issues on the new view
[image]

PR #10615 (2022-06-06): fix(app): Add looking for robots state
# Overview
Fixes looking for robots state on devices landing page
Closes #10483

# Changelog
- adds looking for robots copy
- add spinner icon

# Review requests
Click Refresh button when no robots are found to confirm copy and spinner render

# Risk assessment
Low
commit 00df52d1 (2022-06-06): fix(app): Add looking for robots state (#10615)

* fix(app): Add looking for robots state
Closes #10483
PR #10646 (2022-06-07): fix(app): fix refresh robots spinner spacing and double icon
# Overview
Address 6.0 beta feedback re: refreshing robot spinner spacing and double icon
Closes #10483

# Changelog
- fixed spacing
- removed extra icon
[image]

# Review requests
Click refresh robots button to confirm spacing is fixed and double open in new window icon has been removed.

# Risk assessment
commit 33df1aa4 (2022-06-07): fix(app): fix refresh robots spinner spacing and double icon (#10646)

Closes #10483
closed 2022-06-07 by vabruzzo