Opentrons/opentrons #10484: 6.0 Feedback:  Deck Calibration completed but deck calibrations alert still present

y3rsh, 2022-05-26:
### Overview

Complete deck calibration but alert does not go away.
Must navigate away from the page and back.
[image]
### after navigating away and back
[image]

### Operating system

Mac

reachojhaneha, 2022-05-26:
@koji @y3rsh , i faced the same issue. But after returning to robot settings page from other tabs, this message was gone. More like refresh issue

koji, 2022-05-26:
@reachojhaneha thanks
yup, I just reproduced the issue. I will fix this issue.

closed 2022-05-31 by koji

nusrat813, 2022-06-08:
Verified fixed

PR #10515 (2022-05-31): fix(app): RobotSettings deck calibration banner issue display issue

# Overview
This will close #10484

# Changelog
- Fix deckCalibrationStatus check issue
- Add warning modal for deck calibration (this modal was missing)
- Add polling to solve Banner display issue
- Update a test case and add a new case for warning Banner
- Add a new fixture for test

# Review requests
- After deck calibration, the banner goes away from the RobotSettings Calibration page

# Risk assessment
low
commit 4e975d38 (2022-05-31): fix(app): RobotSettings deck calibration banner issue display issue (#10515)

The implementation for deck calibration wasn't done correctly. This PR fixes that issue and also fix
banner display issue.

fix #10484