Opentrons/opentrons #10485: 6.0 Feedback:  Missing Calibration alert still present after finishing calibration

y3rsh, 2022-05-26:
### Overview

Just finished calibrating the highlighted pipette
[image]

reachojhaneha, 2022-05-26:
I am able to reproduce the issue, looks like refresh issue. After i came to robot settings from device page, calibration details were populated.
[image]
.

closed 2022-06-06 by koji

y3rsh, 2022-06-07:
verified fixed.