Opentrons/opentrons #10487: 6.0 Feedback:  <title> Run protocol button still blue when robot is unavailable

nusrat813, 2022-05-26:
### Current Behavior

Robot is unavailable - run protocol button is still displayed as blue (upon clicking nothing happens)
[image]
[image]

### Expected Behavior

Button should be grayed out 

### Operating system

Mac

jerader, 2022-06-07:
I'm unable to reproduce this on latest edge:

https://user-images.githubusercontent.com/66035149/172452521-50b74795-034b-4b27-89df-8d5d0c176f33.mov

vabruzzo, 2022-06-08:
i am also unable to reproduce

closed 2022-06-14 by y3rsh