Opentrons/opentrons #10490: 6.0 Feedback:  App goes blank while choosing a different tip rack option during tip length calibration

reachojhaneha, 2022-05-26:
### Overview

https://user-images.githubusercontent.com/39049853/170590280-47b7a93d-9c8a-4238-aeda-b93da5cf05b9.mov

### Current Behavior

[image]

There are no network errors

### Expected Behavior

app should not white screen

### Steps To Reproduce

1. Goto devices page
2. click on the robot
3. Goto robot settings
4. Goto Tip length calibrations -> click on the overflow menu
5. click on Recalibrate Tip length and Pipette Offset option
6. Click on "Use a different Tip Rack"
[image]

koji, 2022-05-27:
@reachojhaneha 
Could you try this branch `app_fix-different-tiprack-white-screen` for #10490?

closed 2022-06-01 by koji

nusrat813, 2022-06-07:
Verified fix