Opentrons/opentrons #10493: 6.0 Feedback:  500 error while clicking on "Proceed to Setup" on a /protocol api

reachojhaneha, 2022-05-26:
### Overview

I am getting 500 error, while sending a protocol to robot (not consistently reproducible)

### Current Behavior

[image]
[image]

/protocols response:

```
{
   "errors":[
      {
         "id":"UnexpectedError",
         "title":"Unexpected Internal Error",
         "detail":"sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such table: protocol\n[SQL: SELECT protocol.id \nFROM protocol ORDER BY \"_ROWID_\"]\n(Background on this error at: https://sqlalche.me/e/14/e3q8)",
         "meta":{
            "stacktrace":"Traceback (most recent call last):\n  File \"usr/lib/python3.7/site-packages/sqlalchemy/engine/base.py\", line 1852, in _execute_context\n  File \"usr/lib/python3.7/site-packages/sqlalchemy/engine/base.py\", line 2033, in _handle_dbapi_exception\n  File \"usr/lib/python3.7/site-packages/sqlalchemy/util/compat.py\", line 207, in raise_\n  File \"usr/lib/python3.7/site-packages/sqlalchemy/engine/base.py\", line 1809, in _execute_context\n  File \"usr/lib/python3.7/site-packages/sqlalchemy/engine/default.py\", line 732, in do_execute\nsqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such table: protocol\n[SQL: SELECT protocol.id \nFROM protocol ORDER BY \"_ROWID_\"]\n(Background on this error at: https://sqlalche.me/e/14/e3q8)"
         }
      }
   ]
}

```

### Expected Behavior

I should be able to send the protocol to robot

### Steps To Reproduce

1. Click on device -> go to device details page
2. Select the protocol to send the protocol
3. Click on "Proceed to Setup"
4. /protocols throw 500 error -> and Setup page for protocol is not displayed.

https://user-images.githubusercontent.com/39049853/170591936-9d82c0a6-267b-4932-a8f4-59f71fa85eb7.mov

Protocol Used:

[Test_PUR_Robot_1.json.zip](https://github.com/Opentrons/opentrons/files/8782571/Test_PUR_Robot_1.json.zip)

mcous, 2022-06-10:
Unable to reproduce nor explain how this error could occur. Two theories:

- This particular robot in question had a funny database due to testing of in-development code
- There is a race during database initialization, which would mean subsequent requests would succeed

Given lack of repro, going to close this ticket in favor of re-opening if we are able to reproduce

aaly00, 2022-09-27:
I am running into the same exact issue, and I am not sure how to fix it

aaly00, 2022-09-27:
I am not sure how to reproduce the issue, as it just happened. The only thing is that I tried to run a protocol over ssh. I am also not able to share the protocol, but it's not the cause behind the issue since we were able to run it after I reset the protocols table in ```robot_server.db``` and deleted all protocols from protocols dir under ``` /root/data/opentrons_robot_server```
It seems that there is some kind of corrupted data caused the web server to always issue a 500 Internal server error.