Opentrons/opentrons #10495: 6.0 Feedback:  35 Character renaming of a OT2 displayes outside of <div>

reachojhaneha, 2022-05-26:
### Overview

the robot name and connection symbol should be with in the div

### Current Behavior

[image]

### Steps To Reproduce

1. Goto robot settings -> advanced
2. Rename the robot (35 character)
3. name goes out of the div

koji, 2022-05-27:
@reachojhaneha Did you use `edge`?

[image]

reachojhaneha, 2022-05-27:
@koji , i used alpha
App Software Version
6.0.0-alpha.0

koji, 2022-05-27:
@reachojhaneha thank you for your response.
From the timestamp, 6.0.0-alpha includes a long name issue solution.
As you can see, `edge` doesn't have that issue.

reachojhaneha, 2022-06-08:
Reopening this issue, as this is a regression introduced. This exists in 6.0.1-alpha

[image]

jerader, 2022-06-09:
I am unable to reproduce this on 6.0.0-alpha.1: 

[image]

mikecardarelli, 2022-06-13:
Closing this. We can reopen if this is reproduced somewhere other than Neha's computer

closed 2022-06-13 by mikecardarelli