Opentrons/opentrons #10498: 6.0 Feedback:  Successfully renaming a robot, changes to calibration missing state -> adding a new Unavailable robot to the list

reachojhaneha, 2022-05-26:
### Overview

Rename a robot does the following
-  throws calibration missing banner
- pipette calibration missing

After the calibration  details are rendered correctly after few seconds
- Robot goes to unavailable state
- it is append to the Unavailable list of robots with no heading
- clicking on robot details page - gives the last robot name

### Current Behavior

[image]

### Steps To Reproduce

1. Goto robot settings
2. Rename the robot 

observations:
-  throws calibration missing banner
- pipette calibration missing

After the calibration  details are rendered correctly after few seconds
- Robot goes to unavailable state
- it is append to the Unavailable list of robots with no heading
- clicking on robot details page - gives the last robot name

nusrat813, 2022-06-01:
investigate - @koji 

Discovery client state - new robot that is sharing ip address with "old robot"

y3rsh, 2022-06-14:
verified fixed #10680