Opentrons/opentrons #10501: 6.0 Feedback:  6.0.0-alpha-0

emilywools, 2022-05-27:
### Overview

I updated the robot and afterwards, the app still said that robot has an update

[image]
.
[image]

emilywools, 2022-05-27:
update: when i left the robot detail page, the app no longer shows that an update is available.

emilywools, 2022-05-27:
... but the overflow menu still has an item to "Update robot software"
[image]

ecormany, 2022-05-27:
same as #10041. 

good catch on the overflow menu being affected too, though.

bndo, 2022-06-01:
Closing as duplicate

closed 2022-06-01 by bndo