Opentrons/opentrons #10534: CI: docs build failing with Tutorial in toctree

ecormany, 2022-05-31:
# Overview

Docs build to sandbox is failing when the Tutorial (#10391) is included in the toctree. The failure occurs at the PDF build stage, which is not critical and is blocking HTML deployment.

# Steps to reproduce

Run the [opentrons documentation build](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true) test on a commit on the `docs-tutorial` branch that includes `tutorial` in the toctree in `index.rst`.

# Current behavior

Docs build fails and prevents moving on to the deploy to sandbox stage. There is no obvious error reason in the logs. (The logged warnings happen for other PDFs in the build process and are not fatal.)

```
Run number 1 of rule 'pdflatex'
[459](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:460)
------------
[460](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:461)
------------
[461](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:462)
Running 'pdflatex   -recorder  "OpentronsPythonAPIV2.tex"'
[462](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:463)
------------
[463](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:464)
Latexmk: Index file 'OpentronsPythonAPIV2.idx' was written
[464](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:465)
Latexmk: List of undefined refs and citations:
[465](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:466)
  Reference `new_atomic_commands:v2-atomic-commands' on page 4 undefined on input line 292
[466](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:467)
  Reference `new_complex_commands:v2-complex-commands' on page 5 undefined on input line 300
[467](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:468)
  Reference `new_pipette:new-pipette-models' on page 4 undefined on input line 263
[468](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:469)
  Reference `new_protocol_api:opentrons.protocol_api.contexts.InstrumentContext.transfer' on page 4 undefined on input line 292
[469](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:470)
  Reference `new_protocol_api:opentrons.protocol_api.contexts.ProtocolContext.load_instrument' on page 4 undefined on input line 263
[470](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:471)
  Reference `new_protocol_api:opentrons.protocol_api.contexts.ProtocolContext.load_labware' on page 3 undefined on input line 235
[471](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:472)
  Reference `new_protocol_api:opentrons.protocol_api.labware.Labware.wells' on page 5 undefined on input line 300
[472](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:473)
  Reference `tutorial:tutorial-simulate' on page 1 undefined on input line 96
[459](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:460)
Latexmk: Summary of warnings:
[460](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:461)
  Latex failed to resolve 8 reference(s)
[461](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:462)
Collected error summary (may duplicate other messages):
[462](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:463)
  pdflatex: Command for 'pdflatex' gave return code 256
[463](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:464)
Latexmk: Use the -f option to force complete processing,
[464](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:465)
 unless error was exceeding maximum runs of latex/pdflatex.
[465](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:466)
make[1]: *** [OpentronsPythonAPIV2.pdf] Error 12
[466](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:467)
Makefile:140: recipe for target 'docs/build/pdf/OpentronsPythonAPIV2.pdf' failed
[467](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:468)
rm docs/build/pdf/OpentronsPythonAPIV1.pdf docs/build/html/v1 docs/build/html/v2
[468](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:469)
make: Leaving directory '/home/runner/work/opentrons/opentrons/api'
[469](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:470)
make: *** [docs/build/pdf/OpentronsPythonAPIV2.pdf] Error 2
[470](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:471)
make: unlink: docs/build/html/v1: Is a directory
[471](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:472)
make: unlink: docs/build/html/v2: Is a directory
[472](https://github.com/Opentrons/opentrons/runs/6672763872?check_suite_focus=true#step:8:473)
Error: Process completed with exit code 2.
```

# Expected behavior

Docs build and deploy

# Possible resolutions

1. Go spelunking and find out if there's some LaTeX configuration error that resolves this.
2. `Use the -f option to force complete processing` as suggested in the logs and see if that works.
3. Strip out the PDF build entirely. (I'm in favor of this. Managing a TeX distribution seems like a lot of overhead for little reward.)

laviera, 2022-06-01:
@ecormany - I am going to run this locally

ecormany, 2022-06-03:
Fixed by https://github.com/Opentrons/opentrons/pull/10583

closed 2022-06-03 by ecormany