Opentrons/opentrons #10535: 6.0 Feedback:  Tempdeck slide out menu accepts +/-

nusrat813, 2022-05-31:
### Current Behavior

Temp deck slide-out menu accepts entries "+" "-" and allows users to hit confirm

[image]

nusrat813, 2022-06-01:
Address after beta

koji, 2022-09-10:
Seems that this has been solved.

Still can accept the `+` but the input field can handle it properly.
In terms of `-`, if users type a number, they can add `-` but the input field checks the range and it doesn't allow them to confirm the negative value.

Ex +90 --> 90
      -90 --> show error message