Opentrons/opentrons #10536: 6.0 Feedback:  Dev console error, navigating to section causes "Expected path command" error

bndo, 2022-05-31:
### Overview

In 6.0.0-alpha.0, navigating from Devices to Protocols or Labware results in a console error

### Current Behavior

The minified stack trace (not terribly useful)
[image]

```
Error: <path> attribute d: Expected path command, "…556H8V0H4.44444Z.83L9.17,16.24L1…".
x @ react-dom.production.min.js:23
rl @ react-dom.production.min.js:208
Ou @ react-dom.production.min.js:251
Eu @ react-dom.production.min.js:250
wu @ react-dom.production.min.js:250
xu @ react-dom.production.min.js:250
hu @ react-dom.production.min.js:243
(anonymous) @ react-dom.production.min.js:123
t.unstable_runWithPriority @ scheduler.production.min.js:18
Fi @ react-dom.production.min.js:122
Gi @ react-dom.production.min.js:123
Hi @ react-dom.production.min.js:122
Te @ react-dom.production.min.js:292
Jt @ react-dom.production.min.js:73
```

### Steps To Reproduce

1. launch app and toggle dev tools if you don't have them on.
2. filter the dev tools to Errors only.
3. click on Protocols or Devices or Labware

### Operating system

Mac

### Robot setup or anything else?

no robot connected

bndo, 2022-06-07:
Tested on `6.0.0-alpha.1` 
- Navigating to Devices is clean.
- Navigating to Labware results in one `Error: <path> attribute d: Expected path command...`
- Navigating to Protocols results in three `Error: <path> attribute d: Expected path command...`

b-cooper, 2022-07-01:
Closed By #10990

closed 2022-07-01 by b-cooper

y3rsh, 2022-07-06:
### verified on alpha5
I navigated all over the app and do not see any errors in console.

commit b0f3a047 (2022-07-01): fix(app): remove slideout and modal from robot overflow menu conditional render

Prevent the robot overflow menu from visually persisting above the choose a protocol slideout when
it is opened. Also fix a bad icon svg path that was leaving errors in the console.

Closes #10782, Closes #10536
PR #10990 (2022-07-01): fix(app): remove slideout and modal from robot overflow menu conditional render
# Overview

Prevent the robot overflow menu from visually persisting above the choose a protocol slideout when
it is opened. Also fix a bad icon svg path that was leaving errors in the console.

Closes #10782, Closes #10536

# Changelog

- move the `ChooseProtocolSlideout` and `ConnectionTroubleShooting` modal out of the block that is shown or hidden with the robot overflow menu
- explicitly close the robot overflow menu if either of the above modal or slideout are opened
- fix up a bad svg icon path for `open-in-n [... cut]
commit 884396d9 (2022-07-01): fix(app): remove slideout and modal from robot overflow menu conditional render (#10990)

Prevent the robot overflow menu from visually persisting above the choose a protocol slideout when
it is opened. Also fix a bad icon svg path that was leaving errors in the console.

Closes #10782, Closes #10536