Opentrons/opentrons #10545: 6.0 Feedback:  Delete protocol run record does not in overflow menu does not close after protocol is deleted

nusrat813, 2022-05-31:
### Current Behavior

https://user-images.githubusercontent.com/30632549/171259439-f8cb61c8-8c96-4bdb-9273-5ba63993b044.mp4

Overflow menu for the specific run does not close after the run as been completed 

User can just keep clicking to delete runs for other protocols 

### Expected Behavior

Overflow menu should close after run is deleted 

### Operating system

Mac

y3rsh, 2022-06-03:
I still see dome odd behavior on these menus when I `make -C app dev`
[image]

koji, 2022-06-06:
Closed by #10587

closed 2022-06-06 by koji

y3rsh, 2022-06-07:
verified fixed