Opentrons/opentrons #10553: 6.0 Feedback:  lookup for IP address of an available robot shows as "not found"

reachojhaneha, 2022-05-31:
### Overview

Enter the (wifi)IP of a robot which is available, in "connect to a robot via IP address". Adding a valid IP address shows not found. The robot is connected via usb and adding the wifi IP address shows as not found

### Current Behavior

Uploading Screen Recording 2022-05-31 at 3.38.18 PM.mov…

### Expected Behavior

The wifi IP address should be found

### Steps To Reproduce

1. Connect a device with USB
[image]

2. Goto robot settings and fetch the wifi IP address -> from robot settings
[image]

3. Goto App-settings -> connect to a robot via IP address -> click on set up connection
4. Enter the wifi address -> click on add
[image]

nusrat813, 2022-06-01:
work around - unplug usb 

Investigate - @b-cooper 

Potential fix - read all UP address from discovered robots to back this view

koji, 2022-07-14:
This isn't an issue since the case in this ticket is an edge case.
Basically, users use `connect to a robot via IP address` to detect a robot that isn't in Available robots list.

closed 2022-09-10 by koji