Opentrons/opentrons #10555: 6.0 Feedback:  T1 Protocol Card SVG Fix <title>

mmencarelli, 2022-05-31:
### Overview

This is to fix a spacing issue around the SVG on the protocol cards. 

### Current Behavior

[Link to the figma frame ](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=11331%3A330817)with annotations

[Loom video link](https://www.figma.com/file/2xjSLHXquNsKvC1mV9NMjr/Milestone-2?node-id=955%3A144959) explaining the issue 

- [ ] (13) Think I'll need to work 1-on-1 with a dev to understand what's going on here. Loom video should explain most of it! Just DM me when you start working on this.

aptrons22, 2022-06-24:
Keeping this issue open until further notice from Mel.  Thanks

koji, 2022-07-05:
This will solved by #10993

PR #10993 (2022-07-08): fix(app): Address protocols landing design feedback

# Overview
This will address feedback in #10767 and fix #10208 which is the original design feedback for protocols landing page.
In addition, this will fix #10555

[image]

[width 600px]
[image]

[image]

* in terms of (15), need to get details from Mel. This will be added when she is back.
Opened a new issue  for No.15, #11000

# Changelog
- address design feedback in #10767
closed 2022-07-05 by koji