Opentrons/opentrons #10569: 6.0 Feedback:  Upload a protocol that will parse on the app but will not on the robot.

y3rsh, 2022-06-01:
### Overview

Given the attached protocol which has python 3.10 syntax.
We see that it will upload and be analyzed by the application due to the application having bundled python 3.10.
When we proceed to run nothing will happen because the POST to `/protocols` of the protocol returns 422
[image]

### Current Behavior

Robot selection slide out is still present and you can.t proceed to run.

### Expected Behavior

Not sure what the design is but if the robot cannot parse the protocol we probably should not run the protocol.

### Steps To Reproduce

Use
[310only.py.zip](https://github.com/Opentrons/opentrons/files/8817804/310only.py.zip)
 

### Operating system

Mac

nusrat813, 2022-06-01:
Reuse error state in slide out that was previously used for analysIs errors - https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=12587%3A283630

bndo, 2022-06-23:
Need to surface the caught error

b-cooper, 2022-06-23:
Acceptance criteria for solving this include:

- [ ] display the inline error message from the protocol record creation error response
- [ ] add a spinner/disable to the "proceed to run" CTA after it has been pressed and any of the subsequent requests are in progress

closed 2022-06-28 by vabruzzo

y3rsh, 2022-06-29:
[image]

PR #10925 (2022-06-28): fix(app): handle protocol creation failure in slideouts
# Overview
The app currently shows no error state with the protocol/robot slideout open and a failed protocol run creation from the 'proceed to run' button. This PR adds an error state to the protocol/robot card in the slideout if there is a `runCreationError`. Once another protocol/robot is selected the error is cleared.
I wanted to refactor and lift some state up from the buttons but the location of the `APIHostProvider` prevents this, fortunately there is not much prop drilling involved.

Closes #10569

# Changelog
- adds error state t [... cut]
commit 91c28a93 (2022-06-28): fix(app): handle protocol creation failure in slideouts (#10925)

Closes #10569