Opentrons/opentrons #10593: 6.0 Feedback:  start network discovery as soon as "connect to robot via IP address" slideout opens

reachojhaneha, 2022-06-02:
### Current Behavior

Uploading Screen Recording 2022-06-02 at 2.59.35 PM.mov…

### Expected Behavior

network discovery should start

### Steps To Reproduce

From General settings in app settings
Goto connect to Robot via IP address -> click on "set up connection" button
connect to a robot via IP address slide out is opened
But the network discovery doesn't start

y3rsh, 2022-06-29:
app immediately looks for robots when you click setup connection and reflects this with a spinner.

koji, 2022-06-29:
> app immediately looks for robots when you click setup connection and reflects this with a spinner.

Yes, that is the requirement.
When a user starts the app, it looks for robots 
when a user opens the slideout, the app looks for robots

commit e3025b75 (2022-06-23): fix(app): fix connect robot via ip address functionality and style (#10784)

This will fix the timing of starting discovery mode and conditional rendering of refresh and try
again button. In addition, this PR will fix the styling feedback that is pointed out in #10618
fix #10593 #10594