Opentrons/opentrons #10607: 6.0 Feedback:  App python override path not having any effect

y3rsh, 2022-06-03:
### Overview

When I alter the override python path in the app it seems to have no effect.
I can point it at a legit python version on my machine, or point it at my desktop, the app still seems to point at the bundled python.
I did troubleshooting on it with @TamarZanzouri and we think the [override path may be always getting selected.](https://github.com/Opentrons/opentrons/blob/66a96bc9e76911a3a63b9dae5730607c5ee4cebc/app-shell/src/protocol-analysis/getPythonPath.ts#L26)

mcous, 2022-06-03:
The app isn't trying to pick the first item, nor is the built-in Python the first item when the Python override is set.

I've been unable to reproduce this on my machine using the alpha, though as implemented the app **needs to restart** before the setting will take hold.

Config file (set via app UI):

```js
{
    // ...
    "python": {
        "pathToPythonOverride": "/Users/mc/.pyenv/versions/3.7.12"
    },
    // ...
}
```

App logs

```shell
12:13:42.452 [protocol-analysis/getPythonPath] info: Python environment selected { pythonPath: '/Users/mc/.pyenv/versions/3.7.12/bin/python3' }
```

y3rsh, 2022-06-03:
closed and replaced with #10612

closed 2022-06-03 by y3rsh