Opentrons/opentrons #10618: 6.0 Feedback:  Slide-outs #10168

mmencarelli, 2022-06-03:
### Current Behavior

Link to [annotated frame in Figma ](https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=13793%3A416275)

**Import custom labware def + toast**
- [ ] (1) Size of this button should be 24px (currently 20px)
- [ ] (2) remove the box-shadow
- [ ] (3) Going to sneak this toast edit in here for simplicity sake, but can we add a max-width of these toasts to like 70% of the viewport? This percentage should also work when we shrink down to our min-width size 
- [ ] (4) Can we make it a rule that the toasts stay for 8s instead of 4? Noticing that it's a bit quick. 
- [ ] (5) Could we have the toast fade in and out like[ this example? ](https://polaris.shopify.com/components/feedback-indicators/toast#navigation)
- [ ] (6) Center this text
- [ ] (7) needs 2rem margin-top

[image]
[image]

**Import a protocol slideout, round 2**
- [ ] (1)  center this text
- [ ] (2) remove box-shadow
- [ ] (3) Update the size from 20px to 24px 

**Choose a protocol to run slideout**
- [ ] (1) All links should be our opentrons blue! 
- [ ] (2) Update size of button to 24px, currently 20px
- [ ] (3) Can we add a white background to the SVG? 
- [ ] (4) Have a set-width for the SVG of 76x68px 
- [ ] (5) Noticed that when I hover over a mini card, the cursor doesn't change to the hand vs the mouse? Linking still works, but the cursor doesn't change. 
- [ ] (6) Remove box shadow from .guMWjt

Unfortunately cannot test out the two warning states we have for the mini cards. If possible, would the dev that picks this up be able to send me screenshots? 

[image]
[image]
[image]
[image]

**Choose a robot to run slideout**

- [ ] (1) Add a margin-right: .5rem to the `Slideout_title_Choose` 
- [ ] (2) Make the icon 24px vs 20px 
- [ ] (3) Got an extra 8px margin top above "refresh list". Should be 16px total between the line above and refresh list
- [ ] (4) Missing hover state for the "refresh list" link. Could we change this text to "refresh"? 
- [ ] (5) Robot image should be a fixed width of 64x57px 
- [ ] (6) Technically this content should be vertically centered within the container. So should be down a bit more 
- [ ] (7) Remove `background-color: #f8f8f8;` from the hover state
- [ ] (8) Also missing the cursor change to the hand signifying a clickable item when hovering over this card 
- [ ] (9) Are we nixing this content? if so, let me know so I can update the designs. 
- [ ] (10) Remove box shadow from this button
- [ ] (11) Made a pitch to add a countdown timer when the user clicks "refresh". To the left of the screenshot in figma you'll see the designs 
- [ ] (12) Make the icon bigger, should be 20x20px and should be _darkGreyEnabled_
- [ ] (13) If the discovery times out, then the copy changes to "try again". [Check out the designs on Figma.](https://www.figma.com/file/2xjSLHXquNsKvC1mV9NMjr/Milestone-2?node-id=394%3A115065)

[image]
[image]
[image]

**manually add a robot via IP address slideout**
Full functionality download found on [Figma](https://www.figma.com/file/2xjSLHXquNsKvC1mV9NMjr/Milestone-2?node-id=89%3A57224). 

- [ ] (1) Update icon to 24px 
- [ ] (2) Icon should be fixed to the end of the entire link, vs placed to the right in the same container 
- [ ] (3) Update to be our _label/semibold_ font styling
- [ ] (4) Update this component to our _input/field-with-label_ component. So the height changes to 32px, there's a border-radius, and the states are different. 
- [ ] (5) **Functionality feedback** Upon opening the slideout, the discovery should kick off. Currently does not. 
- [ ] (6) **Functionality feedback** Available robots should automatically float to the top of the list. The order of this last is by last added.
- [ ] (7) Need 8px margin top and bottom on the line 
- [ ] (8) Bump up the type size in this to reflect the label styling 
- [ ] (9) Add in the states to the the _btn-icon/close_
- [ ] (10) Remove the box-shadow from this button 
- [ ] (11) Up the size of the error message, should be 11px 
- [ ] (12) 4px margin between input field and the error label

[image]

**Rename robot slideout**

- [ ] (1) Update icon to 24px 
- [ ] (2) Did we change this copy? if so, let me know and I can update design. 
- [ ] (3) Add a margin of 4px between "35 character max" and error message
- [ ] (4) Remove our box-shadow from the button

[image]

**Factory reset slideout** 

- [ ] (1) Update icon size to 24px 
- [ ] (2) Should be our _toast-basic_ style unless something has changed? 
- [ ] (3) Add 8px margin between each row 
- [ ] (4) Update the checkbox component to the new component style 
- [ ] (5) Update the colour to _darkBlack_
- [ ] (6) Remove box shadow 
[image]

**Labware definition slideout**

- [ ] (1) Missing the states for the 'x' button
- [ ] (2) I'm not 100% sure this is the updated version of this icon? Also, can we center it vertically with the text? Icon should be down a few pixels 
- [ ] (3) Up the size of the icon to 16x16px, currently at 10x10 
- [ ] (4) Add 8px margin between each line
- [ ] (5) Update font colour to _darkGreyEnabled_

[image]

koji, 2022-06-13:
Working on Connect to a robot via IP slideout

koji, 2022-06-13:
Probably I will open PRs separately since that way would make code review easier.

koji, 2022-06-15:
- [ ] Import custom labware def + toast
- [ ] Import a protocol slideout, round 2
- [ ] Choose a protocol to run slideout
- [ ] Choose a robot to run slideout
- [ ] manually add a robot via IP address slideout
- [ ] Rename robot slideout
- [ ] Factory reset slideout
- [ ] Labware definition slideout

PR #10745 (2022-06-15): fix(app): fix long labware name display issue on slideout

# Overview
This PR will fix a long labware apiName display issue on slideout.
fix #10641
Also, fix #10618 partially,  labware definition slideout feedback 

[image]

# Changelog
- Add css property to Flex: `css={{ 'overflow-wrap': 'anywhere' }}`
- Modify style to align with Figma's design

# Review requests
- When selecting `appliedbiosystemsmicroamp_384_wellplate_4
commit 4d465a78 (2022-06-23): style(app): App style design feedback slideout factory reset (#10819)

This PR will fix Factory Slideout design feedback and update checkboxField styling
fix #10618 partially
closed 2022-07-05 by y3rsh