Opentrons/opentrons #10633: Sync color const with Design System Colours

koji, 2022-06-07:
Sync colors
https://www.figma.com/file/l7BAJAGICr10ELsSqQD0Sr?node-id=1780:90941#206886155

koji, 2022-06-07:
- [ ] fundamentals
  - [ ] white
  - [ ] background
  - [ ] backgroundShade
  - [ ] lightBlue
  - [ ] medBlue
  - [ ] focus
 
- [ ] blue
  - [ ] blueEnabled
  - [ ] blueHover
  - [ ] bluePressed
  - [ ] blueSelected (missing on Figma)  

- [ ] darkBlack
  - [ ] darkBlackEnabled
  - [ ] darkBlackHover
  - [ ] darkBlackPressed   (missing on Figma)  
  - [ ] darkBlackSelected
  - [ ] darkBlackLight

- [ ] yellow
  - [ ] dandelionYellowEnabled
  - [ ] dandelionYellowHover
  - [ ] dandelionYellowPressed
  - [ ] dandelionYellowSelected (missing on Figma)  

- [ ] grey
  - [ ] lightGreyEnabled
  - [ ] lightGreyHover
  - [ ] medGreyEnabled
  - [ ] medGreyHover
  - [ ] medGreyPressed
  - [ ] medGreySelected
  - [ ] darkGreyHover
  - [ ] darkGreyPressed
  - [ ] darkGreyDisabled 
 

- [ ] success
  - [ ] successBackground
  - [ ] successEnabled
  - [ ] successHover
  - [ ] successText (missing on Figma)
  - [ ] successDisabled

- [ ] warning
  - [ ] warningBackground
  - [ ] warningEnabled
  - [ ] warningHover
  - [ ] warningText (missing on Figma)
  - [ ] warningDisabled

   
- [ ] error
  - [ ] errorBackground
  - [ ] errorEnabled
  - [ ] errorHover
  - [ ] errorText
  - [ ] errorDisabled

koji, 2022-06-07:
Need?
`export const blueFocus = '#deecff'`

koji, 2022-06-07:
- [ ] StyledText p color={COLOR.darkBlackEnabled} remove `color`

mmencarelli, 2022-06-07:
- focus is in Figma 
- no blueSelected
- we have darkBlackHover
- there is no darkBlackSelected
- we have darkBlackLight
- no dandelionYellowSelected
- we have  lightGreyEnabled and  lightGreyHover
- we have medGreyEnabled
- medGreyPressed and medGreySelected are there and they are the same
-  darkGreyPressed and darkGreyDisabled are there
- successText replaces successHover (was mislabeled)
- warningText replaces warningHover (was mislabeled
- successHover and warningHover get nixed
-

mmencarelli, 2022-06-07:
Also, can we add all the liquid colours? 
[image]

koji, 2022-06-07:
@mmencarelli 
Is it possible to change the name like liquid/blue liquid/orange etc?

I feel using numbers for color names doesn't make code readable.

@sakibh  @smb2268 
Do you have any suggestions on this?

koji, 2022-06-08:
@mmencarelli added colors to `colors.ts`

commit 73312fec (2022-06-08): refactor(app): update colors.ts to align with Design System Colours

There is a difference between code base colors and Design System Colours such as color name and hex
code. This PR will bridge the gap by updating color.ts consts.

fix #10633
PR #10664 (2022-08-08): refactor(app): App style sync color const with colours

# Overview
This will close #10633
Basically, all color names come from the Design System. 
https://www.figma.com/file/l7BAJAGICr10ELsSqQD0Sr/Design-System?node-id=1780%3A90941

**I leave the previous colors on purpose, they will be removed when this PR is ready to merge.**

For example, 
`darkBlack` --> `darkBlackEnabled`  
`errorBG` --> `errorBackground`

keep `darkGrey` since it is used by H/S components. This will be updated in the future.

As Vincent mentioned, we can change names that are vague

# Changelog
- Update colors.ts
- Updat [... cut]
commit 89de0eb1 (2022-08-08): refactor(app): App style sync color const with colours (#10664)

There is a difference between code base colors and Design System Colours such as color names and hex
codes. This PR will bridge the gap by updating color.ts constant.
fix #10633
closed 2022-08-08 by koji