Opentrons/opentrons #10647: 6.0 Feedback:  Module actions created during run pause no not happen immediately.  They seem to happen once the protocol is unpaused.

y3rsh, 2022-06-07:
### Overview

At first I thought the commands were ignored. BUT once I unpaused they all seemed to happen (heard fans spin up and mag module engage)
https://user-images.githubusercontent.com/502770/172441177-e6ff7c8d-1868-418e-bb40-ebe058294f29.mp4

### Current Behavior

Module commands happen when unpaused.

### Expected Behavior

Module commands happen immediately.

### Steps To Reproduce

1. Have modules attached to the robot.  I do not think it matters if they are in the prootocol.
2. Run a protocol
3. Pause the protocol
4. Go to modules tab

mcous, 2022-06-07:
What does the HTTP API report for the module control commands while the protocol is paused? Is the first command in a `running` state or is it in a `queued` state?

y3rsh, 2022-06-07:
@mcous 
I paused the protocol
Then issued the disengage from the mag modules card 
when I get the /runs/RUN_ID/commands
```js
{
    'data': [
        {
            'id': 'commands.LOAD_LABWARE-0',
            'key': 'commands.LOAD_LABWARE-0',
            'commandType': 'loadLabware',
            'createdAt': '2022-06-07T20:40:16.671560+00:00',
            'startedAt': '2022-06-07T20:40:16.671560+00:00',
            'completedAt': '2022-06-07T20:40:16.671560+00:00',
            'status': 'succeeded',
            'params': {
                'location': {'slotName': '1'},
                'loadName': 'opentrons_96_tiprack_300ul',
                'namespace': 'opentrons',
                'version': 1,
                'displayName': '300ul tips'
            }
        },
        {
            'id': 'commands.LOAD_LABWARE-1',
            'key': 'commands.LOAD_LABWARE-1',
            'commandType': 'loadLabware',
            'createdAt': '2022-06-07T20:40:16.861816+00:00',
            'startedAt': '2022-06-07T20:40:16.861816+00:00',
            'completedAt': '2022-06-07T20:40:16.861816+00:00',
            'status': 'succeeded',
            'params': {
                'location': {'slotName': '5'},
                'loadName': 'opentrons_96_tiprack_20ul',
                'namespace': 'opentrons',
                'version': 1,
                'displayName': '20ul tips'
            }
        },
        {
            'id': 'commands.LOAD_PIPETTE-0',
            'key': 'commands.LOAD_PIPETTE-0',
            'commandType': 'loadPipette',
            'createdAt': '2022-06-07T20:40:19.019832+00:00',
            'startedAt': '2022-06-07T20:40:19.019832+00:00',
            'completedAt': '2022-06-07T20:40:19.019832+00:00',
            'status': 'succeeded',
            'params': {'pipetteName': 'p20_single_gen2', 'mount': 'right'}
        },
        {
            'id': 'commands.LOAD_PIPETTE-1',
            'key': 'commands.LOAD_PIPETTE-1',
            'commandType': 'loadPipette',
            'createdAt': '2022-06-07T20:40:21.162072+00:00',
            'startedAt': '2022-06-07T20:40:21.162072+00:00',
            'completedAt': '2022-06-07T20:40:21.162072+00:00',
            'status': 'succeeded',
            'params': {'pipetteName': 'p300_multi_gen2', 'mount': 'left'}
        },
        {
            'id': 'commands.LOAD_MODULE-0',
            'key': 'commands.LOAD_MODULE-0',
            'commandType': 'loadModule',
            'createdAt': '2022-06-07T20:40:21.250903+00:00',
            'startedAt': '2022-06-07T20:40:21.250903+00:00',
            'completedAt': '2022-06-07T20:40:21.250903+00:00',
            'status': 'succeeded',
            'params': {
                'model': 'magneticModuleV2',
                'location': {'slotName': '9'},
[... 190 lines cut ...]
            'key': '3265784f-b51c-4525-80dd-beb451f695c6',
            'commandType': 'magneticModule/disengage',
            'createdAt': '2022-06-07T20:40:41.003270+00:00',
            'startedAt': '2022-06-07T20:40:41.005115+00:00',
            'status': 'running',
            'params': {'moduleId': 'module-0'},
            'intent': 'setup'
        }
    ],
    'links': {
        'current': {
            'href': '/runs/61ddedbc-6690-4c6d-ba36-f13d7ad328dd/commands/3265784f
-b51c-4525-80dd-beb451f695c6',
            'meta': {
                'runId': '61ddedbc-6690-4c6d-ba36-f13d7ad328dd',
                'commandId': '3265784f-b51c-4525-80dd-beb451f695c6',
                'index': 17,
                'key': '3265784f-b51c-4525-80dd-beb451f695c6',
                'createdAt': '2022-06-07T20:40:41.003270+00:00'
            }
        }
    },
    'meta': {'cursor': 0, 'totalLength': 18}
}
```

mcous, 2022-06-08:
Closed by #10865 and #10675

closed 2022-06-08 by mcous

y3rsh, 2022-06-10:
verified, module controls are not available during run paused or running.