Opentrons/opentrons #10658: 6.0 Feedback: device detail page loading process

ecormany, 2022-06-07:
### Overview

Much of the information on the device detail page changes as it is loaded for the first time (or when reloading the page). It flashes warnings that can be distracting or concerning to users. Some information remains inaccurate even after everything is loaded.

### Current Behavior

These sections show incorrect information as they load:

1. Pipettes
     - Pipette images do not show initially
     - "Pipette Offset calibration missing. Calibrate deck first." warning shows very briefly
     - "Pipette Offset calibration needed. Calibrate now" shows slightly longer
2. Protocol runs
    Entries in the Protocol column initially appear as hex identifiers, and only later update to display names.

These sections show incorrect information even after load:
- Protocol run status shows as "Canceled" for all runs prior to 6.0.0-alpha1
- Protocol run duration shows as time between protocol timestamp and present time, instead of between protocol start and end

https://user-images.githubusercontent.com/1958332/172464999-46c2bf77-3cd4-4817-86d1-04f94f5a19d5.mov

### Expected Behavior

- Pipette images should display at same time as pipette names
- Pipette warnings should never show in a state when they are not needed
- Protocol hex identifiers should never be shown instead of names
- Run durations should be accurate

### Steps To Reproduce

Connect to a robot and navigate to its page under Devices. The above behavior happens on first load or when manually triggering a reload from the View menu.

### Operating system

Mac

jerader, 2022-06-21:
Pipette images do not show initially will be addressed in a follow up ticket: https://github.com/Opentrons/opentrons/issues/10847

jerader, 2022-06-21:
Protocol run duration shows as time between protocol timestamp and present time, instead of between protocol start and end will be addressed in this ticket: https://github.com/Opentrons/opentrons/issues/10738)

bndo, 2022-06-23:
Closed by #10848

y3rsh, 2022-06-30:
I have not seen any of the warning banner behavior described but I still see the id taking some seconds to be replaced with the user friendly name.
https://user-images.githubusercontent.com/502770/176688098-50d9a79f-ef87-4d4b-8572-f4d700812208.mp4

closed 2022-06-30 by y3rsh

nusrat813, 2022-07-01:
Verified fixed

commit 872023c5 (2022-06-21): fix(app): pipette card banners and historical protocol run protocolNames render fix

partially closes #10658
commit 3bec9135 (2022-06-22): fix(app): pipette card banners and historical protocol run protocolNames render fix (#10848)

partially closes #10658
PR #10961 (2022-06-30): fix(app): protocol Run Record protocolName no longer renders runId
closes #10899 closes #10658

# Overview

This fixes the bug where the `runid` renders in the protocol name section. Now, if `protocolName` and protocol file are still rendering, `Loading...` will be in place

# Changelog

- change `RecentProtocolRun` and test
- remove unnecessary fetching that I previously added because I thought it fixed this bug

# Review requests

- run a few protocols to get a protocol run record. Refresh the app, the protocol name should say "Loading..." instead of `runId`

# Risk assessment

low
commit 322c71c6 (2022-06-30): fix(app): protocol Run Record protocolName no longer renders runId (#10961)

closes #10899 closes #10658