Opentrons/opentrons #10666: 6.0 Feedback:  UI animation in bottom right corner when downloading run log

rwarshauer, 2022-06-08:
### Overview

When clicking download run log a weird animation is quickly shown in the bottom right corner. Not 100% of the time

https://user-images.githubusercontent.com/106115893/172647048-ec7da7ff-a668-4e03-8474-94f88e3f1f05.mov

.

### Current Behavior

Animation at times

### Expected Behavior

No animation

### Steps To Reproduce

1. Open 6.0.0 App
2. Run a protocol
3. Go to run log
4. After run click download run log
5. observe 

### Operating system

Mac

### Robot setup or anything else?

Dev robot

koji, 2022-06-09:
In this case, the Toast turns off setTimeout (3000ms) and the download process doesn't take so much time. That is the reason why the toast shows up and disappears immediately.

koji, 2022-12-03:
This issue will be solved by https://github.com/Opentrons/opentrons/pull/11797
The related code is updated.

closed 2023-04-15 by koji